espressif/esp-gmf #2: no_codec_player is supported?

Y1hsiaochunnn, 2025-04-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi ,
I recently worked on projects with codec chips such as es8311 and without codec chips, working on esp32-s3. When there is a codec chip on the board, I use the esp_codec_dev component can be easily called, when there is no codec, direct drive way, I defined the i2s channel.

You know, I can test the sound loop with the I2S direct drive board, receive from the microphone and then play from the speaker. In addition, I also tested reading mp3 files from spiffs and playing with esp-audio-player. Everything is fine. Unfortunately, esp-audio-player does not support http mp3 audio streaming.

This library description supports http mp3 audio stream playback, I would like to know if I do not have a codec chip, how should I pay attention to use?

Y1hsiaochunnn, 2025-04-30:
Hi @jason-mao ,
This component works, I managed to get it to play sound through http link, but the sound playback is abnormal, I used a local audio file for testing, sample rate 16000, bit rate 24kbps, mono file.

My function definition looks like this. Is there anything wrong with that?

[image]

[image]

Y1hsiaochunnn, 2025-04-30:
In addition, I initialized i2s with.clk_cfg set to 44100 to output audio data normally. Is this expected?

Y1hsiaochunnn, 2025-04-30:
btw,Is there a component that supports recording audio?

jason-mao, 2025-04-30:
> In addition, I initialized i2s with.clk_cfg set to 44100 to output audio data normally. Is this expected?

Let me first confirm my guess — if you're using I2S at 44100 Hz and it doesn't sound quite right, but 48000 Hz works fine, that’s probably expected.

This is likely because the `simple_player` example has a [default configuration in Kconfig](https://github.com/espressif/esp-gmf/blob/main/examples/esp_audio_simple_player/Kconfig), which forces the output format to 48000 Hz, 2 channels, 16 bits — regardless of the input audio’s sample rate, channel count, or bit depth.

You can change this behavior either by modifying the settings via `menuconfig`, or by configuring the I2S output to match the default format mentioned above.

Also note: the `music_info` parameter in `esp_audio_simple_player_run()` is only needed when the input audio format doesn't contain metadata (e.g., raw PCM or OPUS). For formats like MP3, you don't need to provide it.

Y1hsiaochunnn, 2025-04-30:
I really appreciate your answer,

I also have a little confusion, I saw gmf_ai_audio component, then I used esp32-s3-korvo-2 board to test it, it works fine. Now I want to transplant it, input from the i2s direct drive microphone, and then record. Is there any modification suggestion in this part?

Y1hsiaochunnn, 2025-04-30:
> I really appreciate your answer,
> 
> I also have a little confusion, I saw gmf_ai_audio component, then I used esp32-s3-korvo-2 board to test it, it works fine. Now I want to transplant it, input from the i2s direct drive microphone, and then record. Is there any modification suggestion in this part?

I see, I should refer to the ESP32-S3-EYE audio, they also use digital microphones

Y1hsiaochunnn, 2025-05-30:
Hi @jason-mao ,

There is a problem at the moment, I don't quite understand how to handle the playback logic as a queue task, currently my code when playing http audio (multiple calls to the function) will have http blocking issues and playback exception.

Can you give me some advice?

[image]

jason-mao, 2025-06-05:
> Hi [@jason-mao](https://github.com/jason-mao) ,
>
> There is a problem at the moment, I don't quite understand how to handle the playback logic as a queue task, currently my code when playing http [...]

Hi @Y1hsiaochunnn  
By "multiple calls," do you mean that Task A called play_audio while Task B also called play_audio at the same time?
Could you provide some backtrace and logs? That would be more helpful for me to analyze the issue effectively.

Y1hsiaochunnn, 2025-06-12:
After updating the latest code, this problem will be solved