espressif/esp-gmf #3: Not free the tag field of  esp_gmf_obj_t

kevinkang-Globe, 2025-05-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Many element not free the <tag> field of <esp_gmf_obj_t>. 

For example , need add `esp_gmf_obj_set_tag((esp_gmf_obj_handle_t)file_io, NULL);` to free the <tag>, before free <file_io>

```
static esp_gmf_err_t _file_delete(esp_gmf_io_handle_t io)
{
    file_io_stream_t *file_io = (file_io_stream_t *)io;
    ESP_LOGD(TAG, "Delete, %s-%p", OBJ_GET_TAG(file_io), file_io);
    void *cfg = OBJ_GET_CFG(io);
    if (cfg) {
        esp_gmf_oal_free(cfg);
    }
    esp_gmf_io_deinit(io);
    esp_gmf_obj_set_tag((esp_gmf_obj_handle_t)file_io, NULL);
    esp_gmf_oal_free(file_io);
    
    return ESP_GMF_ERR_OK;
}
```

### Actual behavior (suspected bug)

not free the <tag>, before free <file_io>

```
static esp_gmf_err_t _file_delete(esp_gmf_io_handle_t io)
{
    file_io_stream_t *file_io = (file_io_stream_t *)io;
    ESP_LOGD(TAG, "Delete, %s-%p", OBJ_GET_TAG(file_io), file_io);
    void *cfg = OBJ_GET_CFG(io);
    if (cfg) {
        esp_gmf_oal_free(cfg);
    }
    esp_gmf_io_deinit(io);
    esp_gmf_oal_free(file_io);
    
    return ESP_GMF_ERR_OK;
}
```

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Every time

### Project release version

lastest

### System architecture

other (details in Additional context)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

ZSH

jason-mao, 2025-05-19:
@kevinkang-Globe 

Thank you for raising such an insightful question. There is no memory leak in this case. The `OBJ` object is released through `esp_gmf_obj_delete`, which also frees the associated tag. This is because the tag itself is allocated via `esp_gmf_obj_set_tag`.That said, your proposed usage—calling `esp_gmf_obj_set_tag` during `esp_gmf_io_file_init` to set the tag, and then clearing it in `_file_delete` via `esp_gmf_obj_set_tag(NULL)`—is indeed reasonable. It adheres to the “who allocates, releases” principle and improves readability. I'll consider making this change to enhance clarity and maintainability.

kevinkang-Globe, 2025-05-19:
So before calling `_file_delete` to delete the esp_gmf_io object,  `esp_gmf_obj_delete` has already been called to delete the base object of esp_gmf_io ?

jason-mao, 2025-05-19:
No. The `_file_delete` function is called by `esp_gmf_obj_delete`. `esp_gmf_obj_delete` performs two tasks: it frees the tag and calls the `del_obj` function. The `esp_gmf_io_deinit` does not free the object instance.

kevinkang-Globe, 2025-05-20:
OK, I see. Thanks for your information !

commit 0735ed41 (2025-05-08): Merge branch 'feature/refactor_gmf_folder_structure' into 'main'

Refactor the gmf folder structure

Closes #3

See merge request adf/multimedia/esp-gmf!60
closed 2025-06-05 by kevinkang-Globe