espressif/esp-gmf #5: I encountered an error when trying to play an online MP3 file.

skipify, 2025-05-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I encountered an error when trying to play an online MP3 file.  Since the URL I’m using is an HTTPS address, I modified the http_cfg configuration in audio_simple_player_pool.c of esp_audio_simple_player to add the following line:

`http_cfg.crt_bundle_attach = esp_crt_bundle_attach;`

Even when I switch to an HTTP address, the error still occurs. However, the error message changes to: task_wdt: Task watchdog got triggered.

[6DGDB.mp3.zip](https://github.com/user-attachments/files/20403625/6DGDB.mp3.zip)

### Actual behavior (suspected bug)

esp32-S3R8
esp-idfesp-idf5.3.1

### Error logs or terminal output

```shell
I (23080) XT_PLAYER: play_music_info.url https://localtest.com/a/9c9/16/6DGDB.mp3
I (23090) ESP_GMF_BLOCK: The block buf:0x3c37e068, end:0x3c383068
I (23096) NEW_DATA_BUS: New block buf, num:1, item_cnt:20480, db:0x3c36f1d4
I (23104) ESP_GMF_TASK: Waiting to run... [tsk:http-0x3fcdecb4, wk:0x0, run:0]
I (23111) ESP_GMF_TASK: Waiting to run... [tsk:http-0x3fcdecb4, wk:0x3c36f25c, run:0]
I (23120) ESP_GMF_HTTP: HTTP Open, URI = https://localtest.com/a/9c9/16/6DGDB.mp3

***ERROR*** A stack overflow in task TSK_0x3fcdd6cc has been detected.

Backtrace: 0x40375fc6:0x3fcddac0 0x40380ce5:0x3fcddae0 0x40381dda:0x3fcddb00 0x403837fb:0x3fcddb80 0x40381ea0:0x3fcddba0 0x40381e96:0x00000000 |<-CORRUPTED
--- 0x40375fc6: panic_abort at /Users/admin/.espressif/esp-adf/esp-idf/components/esp_system/panic.c:463
--- 0x40380ce5: esp_system_abort at /Users/admin/.espressif/esp-adf/esp-idf/components/esp_system/port/esp_system_chip.c:92
--- 0x40381dda: vApplicationStackOverflowHook at /Users/admin/.espressif/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:554
--- 0x403837fb: vTaskSwitchContext at /Users/admin/.espressif/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
--- 0x40381ea0: _frxt_dispatch at /Users/admin/.espressif/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
--- 0x40381e96: _frxt_int_exit at /Users/admin/.espressif/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

E (125194) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (125194) task_wdt:  - IDLE0 (CPU 0)
E (125194) task_wdt: Tasks currently running:
E (125194) task_wdt: CPU 0: TSK_0x3fcb2530
E (125194) task_wdt: CPU 1: IDLE1
E (125194) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x420B228B:0x3FCA1DD0 0x420B26AC:0x3FCA1DF0 0x4037827D:0x3FCA1E20 0x42072B06:0x3FCE51C0 0x42072088:0x3FCE5200 0x42071C02:0x3FCE52A0 0x42061E51:0x3FCE5300 0x4205E559:0x3FCE5330 0x4205BBDC:0x3FCE53A0 0x420A8F05:0x3FCE5420 0x420AC416:0x3FCE5450 0x420AC788:0x3FCE54B0 0x403819CD:0x3FCE54E0
--- 0x420b228b: task_wdt_timeout_handling at /Users/admin/.espressif/esp-adf/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
--- 0x420b26ac: task_wdt_isr at /Users/admin/.espressif/esp-adf/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
--- 0x4037827d: _xt_lowint1 at /Users/admin/.espressif/esp-adf/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x42072b06: pvmp3_huffman_parsing at ??:?
--- 0x42072088: pvmp3_framedecoder at ??:?
--- 0x42071c02: esp_mp3_dec_decode at ??:?
--- 0x42061e51: esp_audio_dec_process at ??:?
--- 0x4205e559: esp_audio_simple_dec_process at ??:?
--- 0x4205bbdc: esp_gmf_audio_dec_process at /Users/skipify/wwwroot/new-project/xt_idf2/managed_components/espressif__gmf_audio/esp_gmf_audio_dec.c:148
--- 0x420a8f05: esp_gmf_element_process_running at /Users/skipify/wwwroot/new-project/xt_idf2/managed_components/espressif__gmf_core/src/esp_gmf_element.c:280
--- 0x420ac416: process_func at /Users/skipify/wwwroot/new-project/xt_idf2/managed_components/espressif__gmf_core/src/esp_gmf_task.c:157
--- 0x420ac788: esp_gmf_thread_fun at /Users/skipify/wwwroot/new-project/xt_idf2/managed_components/espressif__gmf_core/src/esp_gmf_task.c:279
--- 0x403819cd: vPortTaskWrapper at /Users/admin/.espressif/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### Steps to reproduce the behavior

I encountered an error when trying to play an online MP3 file. 
I downloaded the MP3 file and placed it on the TF card, and it can be played normally.

### Project release version

0.6

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

sonoma

### Shell

ZSH

jason-mao, 2025-05-26:
@skipify Thanks for your  report, let me check first

skipify, 2025-05-28:
Yes, I modified the audio_simple_player_pool.c file and added the following line after line 21:
http_cfg.out_buf_size = 20 * 1024 * 2;
After this change, the program ran normally.

commit 8ef1aa67 (2025-06-04): Merge branch 'feature/add_gmf_app_utils' into 'main'

Add GMF APP component

Closes #5

See merge request adf/multimedia/esp-gmf!69