espressif/esp-gmf #12: Why not malloc a buffer for esp_gmf_obj's cfg, just like it did for esp_gmf_obj's tag ?

kevinkang-Globe, 2025-06-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

In `esp_gmf_obj_set_config`, malloc a buffer for esp_gmf_obj's cfg, then save the configure in it,
then free this buffer in `esp_gmf_obj_delete`

### Actual behavior (suspected bug)

1.  In  esp-gmf\elements\gmf_ai_audio\examples\aec_rec\main\main.c  `void app_main(void)` pass a local variable `cfg`'s address to `esp_gmf_task_init(&cfg,` &read_task);

```
esp_gmf_task_cfg_t cfg = DEFAULT_ESP_GMF_TASK_CONFIG();
cfg.thread.core = 1;
cfg.thread.stack = 5120;
esp_gmf_task_handle_t read_task = NULL;
esp_gmf_task_init(&cfg, &read_task);
```

2. In  `esp_gmf_task_init` , call `esp_gmf_obj_set_config` save the local variable's address in esp_gmf_obj's `cfg`.

```
int ret = esp_gmf_obj_set_config(obj, cfg, sizeof(*cfg));
ESP_GMF_RET_ON_ERROR(TAG, ret, goto _tsk_init_failed, "Failed set OBJ configuration");
```

If that local variable is invalid, then use this invalid cfg to new an object,  this would cause an error.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Always

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

ZSH

jason-mao, 2025-06-06:
@kevinkang-Globe Thank you very much for your valuable feedback. You've raised a great point. Typically, the configuration should be copied inside esp_gmf_obj_set_config. However, since the cfg content may contain pointers, it cannot be safely freed with a simple free() call — it requires a custom release function. For this reason, we didn't perform a copy within esp_gmf_obj_set_config. That said, your observation about esp_gmf_task_init calling esp_gmf_obj_set_config directly is indeed a bug. We’ll fix this issue accordingly.

kevinkang-Globe, 2025-06-06:
Understood ! Thanks for your confirm.