espressif/esp-gmf #14: I need to get the audio duration, the current playback time, and the ability to play from a specific time position, but I haven't found how to do it.

skipify, 2025-06-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

hi :
I need to get the audio duration, the current playback time, and the ability to play from a specific time position, but I haven't found how to do it.

### Use cases

I hope to provide services similar to the methods described below.
esp_gmf_audio_get_duration
esp_gmf_audio_get_play_time
esp_gmf_audio_seek() (ms)

jason-mao, 2025-06-13:
@skipify @jecelyin  Thank you for your report. I understand that these features are commonly used, but they are not planned for support in the simple player due to its lightweight design. We recommend that users implement these advanced features themselves if needed.

The esp_gmf_audio_get_play_time function is reliable and accurate for tracking playback time. It accumulates the decoded audio data and calculates the playback time based on sample rate, number of channels, and bits per sample. However, features like duration and seek are more complex. To support these with high precision across various audio formats is non-trivial. For stream-based formats such as MP3 or AAC, the difficulty is relatively lower. A typical method is to estimate duration using the formula: duration = file_size / bitrate. To improve accuracy, it's best to subtract the size of metadata such as ID3 tags from the file size. Similarly, seek can be implemented by reversing the calculation: convert a given timestamp into the corresponding file offset, then start reading data from that new position.

We plan to provide these advanced features in a separate media component designed for more complex use cases.

skipify, 2025-06-13:
Thank you very much for your reply. I still hope a unified interface wrapper can be provided as soon as possible to make development more convenient. I’m especially excited to see that your newly open-sourced product "EchoEar" also uses the GMF framework as mentioned in the release notes. Wishing the project continued success and even greater progress!