espressif/esp-gmf #28: Reuse/Idle Pipeline When it is finished

Haerteleric, 2025-09-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

**Description:**
Hi, I’m using **ESP-GMF** to develop an interactive game. My goal is to stitch together a story from multiple MP3 files.

The challenge is that each file has a different runtime, so they may need to start/stop while another file is still playing. I also need to reuse pipelines to play different files once the current one is done (e.g., in a conversation).

Here’s a diagram illustrating my approach:

[image]

### My current GMF architecture:

[image]

* Each track and the mixer have their own GMF-Task assigned.
* They are connected via a ring buffer at the moment.

### Problems encountered:

1. If I start the GMF-Task and one of the track pipelines is not started or has no input URI, there is no sound output at all.
2. If one track pipeline finishes and I try to restart it with a different URI, I get a **Task Timeout** when attempting to restart the pipeline.

### Question:

What is the correct way to:

* Keep pipelines functional even if some don’t have a URI yet? 
* Properly restart a finished pipeline with a new URI without hitting a timeout?

This is my current Soucecode (Work in Progress, Skratch State)
```c
#include "game_audio.h"
#include "utils/pinMap.h"
#include "sdkconfig.h"

#include "esp_gmf_io_codec_dev.h"
#include "esp_gmf_port.h"
#include "esp_gmf_pool.h"
#include "esp_gmf_pool.h"
#include "esp_gmf_err.h"

const char *TAG = "GAME_AUDIO";

#ifndef AUDIO_MANAGER_DEBUG_INFO
    #define AUDIO_MANAGER_DEBUG_INFO(...)
#endif

#ifndef AUDIO_MANAGER_DEBUG_ERROR
    #define AUDIO_MANAGER_DEBUG_ERROR(...)
#endif

#define GAME_AUDIO_EXPECT(func, expect, retVar) \
    do { \
        retVar = func; \
        if (retVar != expect) { \
            AUDIO_MANAGER_DEBUG_ERROR("'%s' failed at %s:%d", #func, __FILE__, __LINE__); \
            goto _exit; \
        } \
    } while (0)

//*-----------------------------GMF Audio IO Init Functions------------------------*/
#include "esp_gmf_io.h"
#include "esp_gmf_io_file.h"
[... 528 lines cut ...]
{
    ESP_LOGI(TAG, "Playing track %d: %s", trackIndex, filePath);
    esp_gmf_pipeline_reset(s_tracks[trackIndex].pipeline);
    esp_gmf_pipeline_set_in_uri(s_tracks[trackIndex].pipeline, filePath);
    esp_gmf_pipeline_run(s_tracks[trackIndex].pipeline);

    if(!s_mixerRunning) {
        esp_gmf_pipeline_run(s_mixer.pipeline);
        s_mixerRunning = true;
    }

    return ESP_GMF_ERR_OK;
}
```

used like this:

```c
    game_audio_init(2);
    game_audio_playTrack(1, "/sdcard/B.mp3"); //(B.mp3 is 29.8 secs)
    game_audio_playTrack(0, "/sdcard/A.mp3"); //(A.mp3 is 14.5 secs)
    vTaskDelay(pdMS_TO_TICKS(15000)); // Wait for a 15 second (A.mp3 is 14.5 secs)
    game_audio_playTrack(0, "/sdcard/A.mp3"); //Restart just to test
    vTaskDelay(pdMS_TO_TICKS(15000)); // Let Audio Play
```

Haerteleric, 2025-09-29:
Got it! Here's a clearer, more polished version of your follow-up that explains the root cause and your fix in a more readable way:

---

I tracked down the cause of the slow audio playback. The issue comes from this line in the mixer implementation:

https://github.com/espressif/esp-gmf/blob/94f7c7058d28fbd848be3893daf9dcbde45b528f/elements/gmf_audio/esp_gmf_mixer.c#L185

The original code causes excessive waiting when only a **single pipeline** is active, which results in **~0.33× playback speed**.

I replaced it with the following logic:

```c
int wait_ticks = ((in_port->wait_ticks < (mixer->frame_time / index)) ? 
                  (mixer->frame_time / index) : 
                  (in_port->wait_ticks));

if (in_port->wait_ticks == 0) {
    wait_ticks = 0;
}
```

With this change, **single-pipeline playback now runs at normal speed**, even when other pipelines are idle.

Haerteleric, 2025-09-29:
yeah i just saw that aswell, but its just the error-exit handler.

Haerteleric, 2025-09-29:
do i need to call `esp_gmf_pipeline_loading_jobs` after `esp_gmf_pipeline_reset`?

Haerteleric, 2025-09-29:
Do i need a new task everytime i restarte the pipeline ? Looking at `esp_gmf_pipeline_reset` The assigned task is reset aswell

TempoTian, 2025-09-29:
As you use file io, when finished you need seek to start to avoid reading data after EOF
```
esp_gmf_err_t esp_gmf_io_set_pos(esp_gmf_io_handle_t handle, uint64_t byte_pos);
```

Haerteleric, 2025-10-27:
@TempoTian  I just got around to playing with the new `audio_render`.

Is there any control over the **Audio Render Processor** pipeline—like element order or configuration? Or do I just have to accept whatever

```c
esp_audio_render_err_t audio_render_proc_create(esp_gmf_pool_handle_t pool, audio_render_proc_handle_t *proc);
```

does?

Also, why is this stuff precompiled anyway? Is there really any IP in simple track mixing?